iamroot12CD/linux #12: 2015년 11월 21일 스터디

norux, 2015-11-21:
오늘의 스터디 토론 내용입니다.

leemchaehoon, 2015-11-21:
디바이스 트리(Device Tree)에 대해 설명이 잘되어 있는  문서를 발견했습니다.
여기서  Android Kernel Hacks.zip 을 다운로드 받고
4장을 보시면 디바이스 트리(Device Tree)에 대해 설명이 잘 나와있습니다.

http://www.kandroid.org/board/board.php?board=AndroidPorting&command=body&no=210

minidump, 2015-11-21:
원문은 여기 http://www.devicetree.org/Device_Tree_Usage

SehongKwon, 2015-11-21:
dtb format 참고

http://iamroot.org/wiki/doku.php?id=%EC%8A%A4%ED%84%B0%EB%94%94:arm_kernel_%EC%8A%A4%ED%84%B0%EB%94%94_10%EC%B0%A8_full_%EB%B0%94%EB%A1%9C%EA%B0%80%EA%B8%B0

slashdeva, 2015-11-21:
http://iamroot.org/wiki/lib/exe/fetch.php?media=%EC%8A%A4%ED%84%B0%EB%94%94:fdt_prop.png

leemchaehoon, 2015-11-21:
http://iamroot.org/wiki/lib/exe/fetch.php?media=%EC%8A%A4%ED%84%B0%EB%94%94:dtb_structure.png

closed 2015-11-28 by norux

leemchaehoon, 2015-12-05:
dtb memory 예제
  memory {
        device_type = "memory";
        reg = <0x0 0x8000000>;  // 128M at 0x0
    };