iamroot12CD/linux #13: fdt_property.nameoff 의 역할

minidump, 2015-11-21:
fdt_property 구조체를 보면 len이 있고 nameoff가 있습니다.
len은 property 키의 value에 해당하는 총 문자열의 길이라고 치면,
과연 nameoff는 무엇인가요?
 
```
struct fdt_property {                                                                             |131         return -FDT_ERR_NOTFOUND;                                                                 
 37         uint32_t tag;                                                                             |132 }                                                                                                 
 38         uint32_t len;                                                                             |133                                                                                                   
 39         uint32_t nameoff;                                                                         |134 int fdt_subnode_offset_namelen(const void *fdt, int offset,                                       
 40         char data[0];                                                                             |135                                const char *name, int namelen)                                     
 41 };  
```

SehongKwon, 2015-11-21:
실제 nameoff에 있는 dt_string의 값은 무엇일까요?

ahnchan2, 2015-11-21:
http://iamroot.org/wiki/lib/exe/fetch.php?media=%EC%8A%A4%ED%84%B0%EB%94%94:dtb_structure.png
링크의 가장 하단부분 표와 같이 node name 과는 다르게 property name 은 별로도 모아 관리 되므로,
해당 property name을 가져오기 위해서는 nameoff 값을 이용하여 property name 을 가져오기 위해
nameoff 변수를 정의한 것 같습니다.

closed 2015-11-30 by norux