iamroot12CD/linux #15: 2015년 11월 28일 스터디

norux, 2015-11-28:


norux, 2015-11-28:
intptr_t 에 대한 설명: http://seoulforest.tistory.com/4

norux, 2015-11-28:
https://msdn.microsoft.com/ko-kr/library/34h23df8.aspx

leemchaehoon, 2015-11-28:
uintptr_t 한글 설명 : http://pr0x1ma.com/archives/46

minidump, 2015-11-28:
[image]
결국 ANSI C표준을 따르기 위해서 (char *)로 바로 바꾸지 않고 (uintptr_t)로 먼저 변환해준게 아닐까요?
```
         namep = (char *)(uintptr_t)fdt_get_name(fdt, nodeoffset, &oldlen);  
```

norux, 2015-11-28:
(void*)(uintptr_t) 추가한 사람의 코멘트: http://lists.denx.de/pipermail/u-boot/2008-August/039070.html

기존에 (void*)로만 되어있었는데, (uintptr_t)를 추가한사람이 있습니다.
그사람이 추가한 이유를 남겨놨습니다.

leemchaehoon, 2015-11-30:
컴파일러가 const 속성을  제거 할때 warnings 이 나올수 있는데.
이를 없애기 위해 uintptr_t 캐스팅을 한번 더 했다 라는것이 되겠네요.

minidump, 2015-12-01:
fdt 헤더 안에는 이미 친절한 주석이 담겨있었어요;; https://github.com/iamroot12D/linux/blob/master/scripts/dtc/libfdt/libfdt.h

leemchaehoon, 2015-12-01:
와 정말 헤더안에 자세한 주석들이 있군요.

closed 2015-12-05 by leemchaehoon