iamroot12CD/linux #16: (struct fdt_property *)가 아니라 (struct fdt_property *)(uintprt_t)로 하는 이유

minidump, 2015-11-28:
아래 return문 안에서 fdt_property 구조체 포인터로 형 변환 하기전에
uintprt_t로 먼저 캐스팅을 하고 있습니다. 이유가 뭘까요?  

```
static inline struct fdt_property *fdt_get_property_w(void *fdt, int nodeoffset,
                                                      const char *name,
                                                      int *lenp)
{
        return (struct fdt_property *)(uintptr_t)
                fdt_get_property(fdt, nodeoffset, name, lenp);
}
```

참고로 uintprt_t는  <uapi/linux/types.h>에서 다음과 같이 재정의 되어 있음
```
typedef unsigned long           uintptr_t;
```

또한 fdt_property 구조체는 /scripts/dtc/libfdt/fdt.h 에 다음과 같이 정의되어 있음
```
struct fdt_property {
        uint32_t tag;
        uint32_t len;
        uint32_t nameoff;
        char data[0];
};
```

fdt_property의 크기는 12바이트라서 8바이트 단위로 정렬(alignment)하기 위해서
uintprt_t로 캐스팅을 하는건지, 아니면 다른 어떤 이유때문인지 궁금합니다.

leemchaehoon, 2015-11-30:
https://github.com/iamroot12D/linux/issues/15
를 참고 하세요.

minidump, 2015-12-01:
https://github.com/iamroot12D/linux/issues/15#issuecomment-160289967 에서 패치 전송자가 남긴 글 확인했습니다.

아래는 그 원문 발췌
```
From: David Gibson <david at gibson.dropbear.id.au>

Enabling -Wcast-qual warnings in dtc shows up a number of places where
we are incorrectly discarding a const qualification.  There are also
some places where we are intentionally discarding the 'const', and we
need an ugly cast through uintptr_t to suppress the warning.  However,
most of these are pretty well isolated with the *_w() functions.  So
in the interests of maximum safety with const qualifications, this
patch enables the warnings and fixes the existing complaints.

Acked-by: Gerald Van Baren <vanbaren at cideas.com>
```

내용처럼 *_w() 함수들은 const가 없어서 그냥 (void *)로 1차 캐스팅만해도 문제 없지만
그렇지 않은 함수들에서는 경고들이 다수 존재해서 이를 제거하기 위해 
명시적(다소 ugly하지만)으로 2차 캐스팅(uintptr_t)로 해준것이었군요.

그리고 `-Wcast-qual` 옵션은 패치 서미터가 테스트하려고 적용한것 같아요. (Makefile에는 없음)

closed 2015-12-01 by minidump