iamroot12CD/linux #17: 33주차 2015년 12월 05일 스터디

leemchaehoon, 2015-12-05:
33주차 스터디 시작합니다.

norux, 2015-12-05:
메모리 뱅크?? http://old.lameproof.com/398611

norux, 2015-12-05:
initrd : http://www.ylabs.co.kr/index.php?document_srl=1493&mid=board_centos

yhkoo, 2015-12-05:
http://www.iamroot.org/xe/Kernel_8_ARM/61099

leemchaehoon, 2015-12-05:
dtb memory 예제
 memory {
  device_type = "memory";
  reg = <0x0 0x8000000>; // 128M at 0x0
 };

norux, 2015-12-05:
http://www.iamroot.org/xe/Kernel_10_ARM/178300

ahnchan2, 2015-12-05:
/*   

     * Make sure that the DTB doesn't end up in the final

     * kernel's .bss area. To do so, we adjust the decompressed

     * kernel size to compensate if that .bss size is larger

     * than the relocated code.

     */

    ldr r5, =_kernel_bss_size @@ r5 = 압축되지 않은 vmlinux의 bss 영역의 크기 

    adr r1, wont_overwrite    @@ r1 = wont_overwrite:의 주소 

    sub r1, r6, r1

    subs  r1, r5, r1

    addhi r9, r9, r1

위 코드에 대한 설명

http://www.iamroot.org/xe/Kernel_10_ARM/179812

norux, 2015-12-06:
흥미로운 이력을 하나 발견했습니다.
2011년에 atags_to_fdt.c 파일을 작성한 사람의 커밋입니다.
https://github.com/raspberrypi/linux/commit/b90b9a38251e9c89c34179eccde57411ceb5f1aa

> #### ARM: zImage: allow supplementing appended DTB with traditional ATAG data

>> #### ARM: zImage: 전통적인 ATAG 데이터를 가진 appended DTB를 지원합니다.

> Some old bootloaders can't be updated to a device tree capable one,
yet they provide ATAGs with memory configuration, the ramdisk address,
the kernel cmdline string, etc.  To allow a device tree enabled
kernel to be used with such bootloaders, it is necessary to convert those
ATAGs into FDT properties and fold them into the DTB appended to zImage.

>> **일부 오래된 부트로더는 디바이스 트리를 다룰 수 있는 형식으로 업데이트해주지 않고, 아직도 메모리 설정, 램디스크 주소, 커널의 커맨드라인 명령어 등을 ATAG를 통해 제공하고 있습니다. 이러한 부트로더를 사용하는 커널에 디바이스트리를 사용할 수 있게하기 위해서 ATAG데이터를 FDT의 프로퍼티로 변환하여 추가된 DTB를 zImage안에 넣어주여야 할 필요가 있습니다.**

> Currently the following ATAGs are converted:

현재 변환되는 ATAG데이터는 다음과 같습니다.

> ```
	ATAG_CMDLINE
	ATAG_MEM
	ATAG_INITRD2
```

>If the corresponding information already exists in the appended DTB, it
is replaced, otherwise the required node is created to hold it.

>> **추가된 DTB안에 이미 위의 정보들이 존재한다면 데이터를 대체하고, 존재하지 않는다면 새로 노드를 생성하여 담습니다.**

> The code looks for ATAGs at the location pointed by the value of r2 upon
entry into the zImage code.  If no ATAGs are found there, an attempt at
finding ATAGs at the typical 0x100 offset from start of RAM is made.
Otherwise the DTB is left unchanged.

>> **코드는 ATAG 데이터를 zImage코드 영역에 있는 r2 레지스터가 가리키는 위치에서 찾습니다. 만약 ATAG를 찾을 수 없다면, 전형적인 ATAG위치인 메모리의 시작주소로부터의 0x100 오프셋 위치에서 찾습니다. 이후에도 실패한다면, DTB를 수정하지 않습니다.**

> This started from an older patch from John Bonesio <bones@secretlab.ca>,
with contributions from David Brown <davidb@codeaurora.org>.

> Signed-off-by: Nicolas Pitre <nicolas.pitre@linaro.org>
Tested-by: Shawn Guo <shawn.guo@linaro.org>
Tested-by: Dave Martin <dave.martin@linaro.org>
Tested-by: Thomas Abraham <thomas.abraham@linaro.org>

norux, 2015-12-06:
그리고 위 코드에대해 64비트 지원부분을 추가한 이력입니다.

gregory.clement@free-electrons.com 이 아저씨한테 64비트시, sizeof(mem_reg_property)/4 이 코드에 문제가 없는지 물어봐야할것 같습니당 :)

https://github.com/raspberrypi/linux/commit/faefd550c45d8d314e8f260f21565320355c947f

> #### ARM: 7722/1: zImage: Convert 32bits memory size and address from ATAG…
… to 64bits DTB

>When CONFIG_ARM_APPENDED_DTB is selected, if the bootloader provides
an ATAG_MEM it replaces the memory size and the memory address in the
memory node of the device tree. In the case of a system which can
handle more than 4GB, the memory node cell size is 4: each data
(memory size and memory address) are 64 bits and then use 2 cells.

>The current code in atags_to_fdt.c made the assumption of a cell size
of 2 (one cell for the memory size and one cell for the memory
address), this leads to an improper write of the data and ends with a
boot hang.

>This patch writes the memory size and the memory address on the memory
node in the device tree depending of the size of the memory node (32
bits or 64 bits).

>It has been tested in the 2 cases:
- with a dtb using skeleton.dtsi
- and with a dtb using skeleton64.dtsi

>Signed-off-by: Gregory CLEMENT <gregory.clement@free-electrons.com>
Acked-by: Nicolas Pitre <nico@linaro.org>

번역은 딱히안해 될듯... 걍 64비트 추가했다는 내용이네요

closed 2015-12-12 by minidump