iamroot12CD/linux #19: 34주차 2015년 12월 12일 스터디

minidump, 2015-12-12:


minidump, 2015-12-12:
지난시간 리뷰 끝, 10분 쉬고 갑니다.

다음 시간에는 head.S의 dtb_check_done 직전의 코드 블록이 정확히 이해가 안된상태로 넘어가서
wiki 페이지에 정리된 내용을 다시 읽고 넘어가기로 했습니다.
* [decompress/head.S 분석(12차A조)](http://iamroot.org/wiki/doku.php?id=kernel_original_12_a:%EB%B6%80%ED%8A%B8_%EA%B4%80%EB%A0%A8#중복_영역_검사를_하여_중복되지_않았다고_판단하면_wont_overwrite_decompress_준비)

minidump, 2015-12-12:
아래 참고해서 미진했던 코드블록 분석중
http://www.iamroot.org/xe/Kernel_10_ARM/179812

leemchaehoon, 2015-12-13:
dtb_check_done: 라벨 까지 메모리 구조

```c
/*
 *   +-----------------------+  <- r10
 *   | buffer(64K),heap 영역 |     (848라인) add r10, sp, #0x10000
 *   +-----------------------+  <- sp
 *   |        stack          |
 *   +-----------------------+
 *   |         bss           |
 *   +-----------------------+  <- r6
 *   |                       |
 *   |        D  T  B        |  r5 = 갱신된 DTB size
 *   |                       |
 *   |Magic|TotalSize|.. |.. |
 *   +-----------------------+  <- DTB 시작
 *   |                       |
 *   |                       |
 *   |                       |  - 해결되지 않은 의문점
 *   |      K e r n e l      |    _kernal_bss_size 가 zImage 보다 작을 경우,
 *   |                       |    DTB 가 복사될 위치가 _kernel_bss 마지막 영역과
 *   |      (compressed)     |    일치하지 않게 되는데, 이 때의 처리가 아직없다.
 *   |                       |
 *   |                       |
 *   +-----------------------+
 *   |      Page Table       |
 *   +-----+-----------------+
 *   |     |      Atag       |
 *   |     +-----------------+
 *   |                       |
 *   +-----------------------+
*/
```

closed 2016-01-17 by leemchaehoon