iamroot12CD/linux #22: 35주차 2015년 12월 19일 스터디

minidump, 2015-12-19:
dtb_check_done: 이후부터 시작합니다.

minidump, 2015-12-19:
Documentation/HOWTO의 한글 번역판,
스터디 들어가기 전에 사실 이것부터 읽고 시작했다면 싶을 정도로 좋은 내용이 많습니다.
꼭 읽어보세요.
https://github.com/torvalds/linux/blob/master/Documentation/ko_KR/HOWTO

winds02, 2015-12-19:
LC0, GOT 등에 대해 조금 더 알아보고 가면 좋을 것 같습니다.
11차 A팀 분석 내용을 참조로.

https://e4deen.wordpress.com/2014/06/21/39/

rilts, 2015-12-19:
wont_overwrite 에서의 메모리 상태가 잘 정리 되어 있는 거 같습ㄴ니다.
http://www.iamroot.org/xe/FreeBoard/206769

norux, 2015-12-19:
http://iamroot.org/wiki/doku.php?id=kernel_original_12_a:부트_관련

ahnchan2, 2015-12-22:
커널 압축해제 과정의 메모리 맵
[image]

leemchaehoon, 2016-01-16:
cache and flush 레지스터 상태 값  변경 flow

[cache_flush_레지스터 상태값.xlsx](https://github.com/iamroot12D/linux/files/92885/cache_flush_.xlsx)
[cache_flush_레지스터 상태값.ods.zip](https://github.com/iamroot12D/linux/files/92886/cache_flush_.ods.zip)

closed 2016-01-17 by leemchaehoon