iamroot12CD/linux #24: 36주차 2016년 1월 9일 스터디

minidump, 2016-01-09:
새해 복 많이 받으세요.
36주차 스터디 시작합니다.
장소는 토즈 강남점 3F 8번방(파고다 뒷 건물)

minidump, 2016-01-09:
두 번째 restart부터 시작.
call_cache_fn으로 호출할 당시 r3에 16(오프셋)을 집어넣어서 proc_types의 cache flush를 참조함.
과거 분석할 당시의 첫번째 restart에서는 cache on을 위해 8을 사용했었음.

```
 * Table for cache operations.  This is basically:
 *   - CPU ID match
 *   - CPU ID mask
 *   - 'cache on' method instruction
 *   - 'cache off' method instruction
 *   - 'cache flush' method instruction
```

SehongKwon, 2016-01-09:
.align 의 directive는 
align을 맞추기 위하여 해당영역이 code영역일시nop를 채우고
그외의 영역일시 0을 채운다고 합니다. (아래 링크)
http://stackoverflow.com/questions/14544617/what-does-the-align-directive-do

minidump, 2016-01-09:
```
.align n
```
n은 2의 n승을 말하고, 
바로 다음 인스트럭션의 주소 체계에만 반영되는것으로 확인되었습니다.
(이하 인스트럭션이 모두 해당되는것이 아님)
(만약 주소가 2의 n승으로 딱 정렬되지 않는다면 그 전까지는 메모리상에서 0x0으로 채워지는걸 확인)

minidump, 2016-01-09:
분석중에 나온 W()는 /arch/arm/include/asm/unified.h 에서 정의되어 있습니다.
``` C
#define ARM(x...)
#define THUMB(x...)	x
#ifdef __ASSEMBLY__
#define W(instr)	instr.w
#define BSYM(sym)	sym + 1
#else
#define WASM(instr)	#instr ".w"
#endif
```

leemchaehoon, 2016-01-09:
ARM align 시뮬 결과
[image]

SehongKwon, 2016-01-09:
[image]

minidump, 2016-01-09:
오랜만에 컴구관련 용어가 나와서 아래 참고
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/3738.html
하버드 아키텍처는 데이터와 명령어 버스가 구분되어 있는 구조,
폰노이만 아키텍처는 데이터와 명령어를 땡겨오는 버스를 하나만 가지고 있는 구조

rilts, 2016-01-09:
http://forum.falinux.com/zbxe/index.php?document_srl=534002&mid=Kernel_API

slashdeva, 2016-01-09:
clidr
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0464d/BABHCIHB.html

SehongKwon, 2016-01-09:
ARM asm 
CLZ 명령어 설명
http://forum.falinux.com/zbxe/index.php?document_srl=574399&mid=lecture_tip

SehongKwon, 2016-01-09:
clidr의 ctype의미
[image]

winds02, 2016-01-09:
10차에서 분석 내용.
http://www.iamroot.org/xe/Kernel_10_ARM/180153

SehongKwon, 2016-01-09:
라즈베리파이2 arm reference 매뉴얼 위치
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/

SehongKwon, 2016-01-09:
l1 harvard ? hierarchy?
[image]

winds02, 2016-01-09:
결론적으로,

Harvard 구조와 Hierarchical은 관계가 없다.

 - Harvard 구조는 Instruction cache와 Data cache가 분리되어 있는 구조를 뜻할 뿐이고,
 - Hierarchical은 아래와 같이 멀티 레벨 계층을 의미한다.
   L1 cache >> L2 cache
   L2 cache << L1 cache