iamroot12CD/linux #25: 39주차 2016년 1월 16일 스터디

elinjkim, 2016-01-16:


leemchaehoon, 2016-01-16:
[cache_flush_레지스터 상태값.xlsx](https://github.com/iamroot12D/linux/files/92807/cache_flush_.xlsx)

norux, 2016-01-16:
[cache_flush_routine.ods.zip](https://github.com/iamroot12D/linux/files/92808/cache_flush_routine.ods.zip)

leemchaehoon, 2016-01-16:
```
2802                  * clidr : cache level id register                              
2803                  * Figure 4.19. CLIDR bit assignments                           
2804                  *  29 27  24    21     18     15     12      9      6      3      0
2805                  *+-----------------------------------------------------------------+
2806                  *| |LoUU|Loc|LoUIS|Ctype7|Ctype6|Ctype5|Ctype4|Ctype3|Ctype2|Ctype1|
2807                  *+-----------------------------------------------------------------+
2808                  * Ctype1 : Level 1 cache의 구현 type                           
2809                  * Ctype2 : Level 2 Cache의 구현 type       
```

minidump, 2016-01-16:
http://www.element14.com/community/servlet/JiveServlet/previewBody/17030-102-1-52869/ARM%20System%20Developers%20Guide-Designing%20and%20Optimizing%20System%20Software.pdf

p413 참고

norux, 2016-01-16:
[image]

leemchaehoon, 2016-01-16:
```
 * L2 캐시 지원 여부를 알아아보는 소스                                          
 * http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0464d/BABHCIHB.html
 * clidr[26:24] LoC                                                             
 *       Indicates the Level of Coherency for the cache hierarchy:              
 *       - 0b000 L2 cache not implemented on the processor.                         
 *       - 0b010 L2 cache coherency.                                            
 * r3 = clidr & 0x07000000;                                                     
 * r3 = r3 >> 23;                                                               
 * if(r3 == 0)  // L2 캐시 지원하지않음.                                        
 *      goto finished;                
```

elinjkim, 2016-01-16:
# Thrashing : 캐시 메모리 안에 동일한 위치에 대해 소프트웨어 병목 현상이 발생하는 것
Direct mapped cache에서는 메인 메모리로부터 값을 저장하기 위해 사용할 수 있는 공간이 하나이므로 가장 높은 thrashing을 야기할 수 있음
=> thrashing 때문에 캐시 라인을 반복해서 제거하고 로딩하는 작업이 일어남

* cache thrashing : 실행하지 않는 루틴을 연속적으로 제거할 때 캐시미스가 반복되는 현상

=> trashing(스레싱)을 줄이기위해서 캐시는 way라는 더 작은 단위로 쪼개짐   .
하나의 메인메모리 주소에 대해 캐시 안에 여러개의 저장 위치를 제공할 수 있음(세트 연상 캐시, set associativity)

## 12.2.4. Set Associativity
- 목표 : 스레싱의 주기를 줄이기 위함
- way : 캐시메모리를 동일한 크기로 나눠놓은 단위
기존 256라인의 하나의 way  ->  64라인의 4개의 way로 사용
- Set index : 동인한 세트 인덱스를 가지는 4개의 캐시라인을 동일한 세트안에 있다고 말함
- Set Associativity : 세트 인덱스가 가리키는 캐시라인의 세트

- 메인메모리의 특정 위치로부터의 데이터나 코드 블록은 한 세트의 메모리인 어떤 캐시라인 안에 저장됨
- 세트안에 값을 저장하는 것은 동일한 코드나 데이터 블록을 한 세트안에 2개의 캐시 라인을 동시에 위치시키지 못하게 하기 위해 배타적으로 동작

# 4-way 세트연상 (위의 그림)

* 메인 메모리의 어떤 하나의 위치는 캐시안에 4개의 다른 위치에 매핑

- 태그에 대한 필드가 2비트 더 큼
- 세트 인덱스 비트 필드는 2비트 더 작음

norux, 2016-01-16:
[image]
[image]

elinjkim, 2016-01-16:
## 12.5.4. Cleaning The D-Cache Using Way and Set Index
- WAY 및 세트 인덱스 주소를 사용해서 D-캐시 클린하기 
 (코어는 해당 way와 set index address를 가지고 각 캐시라인을 선택)
- ARM 코어에서는 캐시안에 해당 위치를 찾기 위해 way와 set index를 사용하여 하나의 캐시라인을 clean하고 flush를 수행 함
 (일반적으로 "way를 선택하는 영역 + way 안에 세트를 선택하는 영역"의 두 필드를 포함)
- 레지스터가 생성되면 Rd를 CP15:c7 레지스터로 이동하기 위해 적절한 MCR(coprocessor -> register)를 사용

> way를 이용하여 캐시라인을 클린하고 플러시하는 데 사용하는 MRC명령어 (위의 표)
- 명령어 캐시라인 플러시
- 데이터 캐시라인 플러시
- 데이터 캐시라인 클린 
- 데이터 캐시라인 클린 및 플러시

SehongKwon, 2016-01-16:
Table 4-47 shows the individual bit field and complete register encodings for the CCSIDR. The
CSSELR determines which CCSIDR to select.

결국 cssidr(캐시의 way나 라인수)를 읽어오기 전에 어떤 level의 캐시를 읽어올지 선택을 csselr로 해주는 것이라고 되어있습니다.

norux, 2016-01-16:
https://ko.wikipedia.org/wiki/캐시_일관성

SehongKwon, 2016-01-16:
https://en.wikipedia.org/wiki/ARM_Cortex-A7
라즈베리 파이가 사용하는 코어의 cache정보를 찾아봤습니다.

L1 cache	8–64 KB/8–64 KB                    <-- L1은 8 way 64KB 입니다.
(32KB L1 instruction cache and a 32KB L1 )
L2 cache	Optional, up to 1 MB               <-- L2는  512kb입니다.
( 2836 has a 512KB L2 cache)

https://www.raspberrypi.org/forums/viewtopic.php?t=98428

leemchaehoon, 2016-01-17:
cache_flush 레지스터 흐름

[cache_flush_레지스터 상태값.xlsx](https://github.com/iamroot12D/linux/files/93205/cache_flush_.xlsx)

[cache_flush_레지스터 상태값.ods.zip](https://github.com/iamroot12D/linux/files/93206/cache_flush_.ods.zip)