iamroot12CD/linux #30: 41주차 2016년 1월 30일 스터디

rilts, 2016-01-30:


ahnchan2, 2016-01-30:
[ARM중] kernel/head.S PAGE_OFFSET 과 TEXT_OFFSET

http://www.iamroot.org/xe/Kernel_8_ARM/61269

leemchaehoon, 2016-01-30:
라즈베리 파이2 기본 .config 파일
[raspberrypi2_config.txt](https://github.com/iamroot12D/linux/files/110703/raspberrypi2_config.txt)

ahnchan2, 2016-01-30:
ARM Mode와 PSR..너희들은 누구냐?

http://egloos.zum.com/recipes/v/5618965

rilts, 2016-01-30:
https://e4deen.wordpress.com/2014/07/12/

leemchaehoon, 2016-01-30:
atags header, atags core 

linux/arch/arm/include/uapi/asm/setup.h
```asm
/* The list ends with an ATAG_NONE node. */                                     
#define ATAG_NONE       0x00000000                                              
                                                                                
struct tag_header {                                                             
        __u32 size;                                                             
        __u32 tag;                                                              
};                                                                              
                                                                                
/* The list must start with an ATAG_CORE node */                                
#define ATAG_CORE       0x54410001                                              
                                                                                
struct tag_core {                                                               
        __u32 flags;            /* bit 0 = read-only */                         
        __u32 pagesize;                                                         
        __u32 rootdev;                                                          
}; 
```

rilts, 2016-01-30:
http://www.iamroot.org/xe/index.php?mid=Kernel_10_ARM&page=2&document_srl=181691