iamroot12CD/linux #31: 43주차 2016년 2월 13일 스터디

rilts, 2016-02-13:
43주차 스터디를 시작합니다.

leemchaehoon, 2016-02-13:
http://forum.falinux.com/zbxe/index.php?comment_srl=636468&listStyle=gallery&mid=lecture_tip&sort_index=last_update&order_type=asc&page=69&document_srl=785092

SCU에 관한 이야기

norux, 2016-02-13:
https://github.com/raspberrypi/linux/commit/bc41b8724f24b9a27d1dcc6c974b8f686b38d554

rilts, 2016-02-13:
__fixup_smp A팀 정리 내용이네요.

http://jake.dothome.co.kr/__fixup_smp/

leemchaehoon, 2016-02-13:
http://stackcanary.com/?p=622

about alt.smp

rilts, 2016-02-13:
__do_fixup_smp_on_up:
분석 그림 올려 드려요.
[image]

SehongKwon, 2016-02-13:
분석하지 않고 넘어간 pv_table 에 관한 링크들 입니다.

http://stackcanary.com/?p=616
http://jake.dothome.co.kr/__fixup_pv_table/

billpark, 2016-02-13:
ENTRY 와 ENDPROC

http://stackcanary.com/?p=485

leemchaehoon, 2016-02-15:
라즈베리 파이2 vmlinux 헤더 정보

```
$ readelf --headers vmlinux
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x80008000
  Start of program headers:          52 (bytes into file)
  Start of section headers:          8839680 (bytes into file)
  Flags:                             0x5000202, has entry point, Version5 EABI, soft-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         6
  Size of section headers:           40 (bytes)
  Number of section headers:         35
  Section header string table index: 32

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .head.text        PROGBITS        80008000 008000 000214 00  AX  0   0  4
  [ 2] .text             PROGBITS        80008240 008240 55816c 00  AX  0   0 64
  [ 3] .fixup            PROGBITS        805603ac 5603ac 000028 00  AX  0   0  4
  [ 4] .rodata           PROGBITS        80561000 561000 16464c 00   A  0   0 64
  [ 5] __bug_table       PROGBITS        806c5650 6c5650 006660 00   A  0   0  1
  [ 6] __ksymtab         PROGBITS        806cbcb0 6cbcb0 007100 00   A  0   0  4
  [ 7] __ksymtab_gpl     PROGBITS        806d2db0 6d2db0 0055b8 00   A  0   0  4
  [ 8] __kcrctab         PROGBITS        806d8368 6d8368 003880 00   A  0   0  4
  [ 9] __kcrctab_gpl     PROGBITS        806dbbe8 6dbbe8 002adc 00   A  0   0  4
  [10] __ksymtab_strings PROGBITS        806de6c4 6de6c4 01c938 00   A  0   0  1
  [11] __param           PROGBITS        806faffc 6faffc 000ef0 00   A  0   0  4
  [12] __modver          PROGBITS        806fbeec 6fbeec 000114 00   A  0   0  4
  [13] __ex_table        PROGBITS        806fc000 6fc000 001020 00   A  0   0  8
  [14] .ARM.unwind_idx   ARM_EXIDX       806fd020 6fd020 02ff20 00  AL 19   0  4
  [15] .ARM.unwind_tab   PROGBITS        8072cf40 72cf40 047d08 00   A  0   0  4
  [16] .notes            NOTE            80774c48 774c48 000024 00  AX  0   0  4
  [17] .vectors          PROGBITS        00000000 778000 000020 00  AX  0   0  4
  [18] .stubs            PROGBITS        00001000 779000 0002c0 00  AX  0   0 32
  [19] .init.text        PROGBITS        807752e0 77d2e0 03c558 00  AX  0   0 32
  [20] .exit.text        PROGBITS        807b1838 7b9838 001578 00  AX  0   0  4
  [21] .init.proc.info   PROGBITS        807b2db0 7badb0 00023c 00   A  0   0  1
  [22] .init.arch.info   PROGBITS        807b2fec 7bafec 0000c8 00   A  0   0  4
  [23] .init.tagtable    PROGBITS        807b30b4 7bb0b4 000048 00   A  0   0  4
  [24] .init.smpalt      PROGBITS        807b30fc 7bb0fc 0064d0 00   A  0   0  4
  [25] .init.data        PROGBITS        807b95e0 7c15e0 01cda4 00  WA  0   0 32
  [26] .data..percpu     PROGBITS        807d7000 7df000 005080 00  WA  0   0 64
  [27] .data             PROGBITS        807de000 7e6000 085af4 00  WA  0   0 64
  [28] .data..page_align PROGBITS        80864000 86c000 002000 00  WA  0   0 4096
  [29] .bss              NOBITS          80866000 86e000 0bd78c 00  WA  0   0 64
  [30] .comment          PROGBITS        00000000 86e000 00005b 01  MS  0   0  1
  [31] .ARM.attributes   ARM_ATTRIBUTES  00000000 86e05b 00002f 00      0   0  1
  [32] .shstrtab         STRTAB          00000000 86e08a 000173 00      0   0  1
  [33] .symtab           SYMTAB          00000000 86e778 1f1520 10     34 100441  4
  [34] .strtab           STRTAB          00000000 a5fc98 16f14d 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings)
  I (info), L (link order), G (group), T (TLS), E (exclude), x (unknown)
  O (extra OS processing required) o (OS specific), p (processor specific)

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x008000 0x80008000 0x80008000 0x76cc6c 0x76cc6c R E 0x8000
  LOAD           0x778000 0x00000000 0x80775000 0x00020 0x00020 R E 0x8000
  LOAD           0x779000 0x00001000 0x80775020 0x002c0 0x002c0 R E 0x8000
  LOAD           0x77d2e0 0x807752e0 0x807752e0 0xf0d20 0x1ae4ac RWE 0x8000
  NOTE           0x774c48 0x80774c48 0x80774c48 0x00024 0x00024 R E 0x4
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RWE 0x10

 Section to Segment mapping:
  Segment Sections...
   00     .head.text .text .fixup .rodata __bug_table __ksymtab __ksymtab_gpl __kcrctab __kcrctab_gpl __ksymtab_strings __param __modver __ex_table .ARM.unwind_idx .ARM.unwind_tab .notes 
   01     .vectors 
   02     .stubs 
   03     .init.text .exit.text .init.proc.info .init.arch.info .init.tagtable .init.smpalt .init.data .data..percpu .data .data..page_aligned .bss 
   04     .notes 
   05     

```

leemchaehoon, 2016-02-20:
```
arch/arm/mm/proc-v7.S
629         /*                                                                      
630          * Match any ARMv7 processor core.                                      
631          */                                                                     
632         .type   __v7_proc_info, #object                                         
633 __v7_proc_info:                                                                 
634         .long   0x000f0000              @ Required ID value                     
635         .long   0x000f0000              @ Mask for ID                           
636         __v7_proc __v7_proc_info, __v7_setup                                    
637         .size   __v7_proc_info, . - __v7_proc_info         

arch/arm/mm/proc-v7.S
489 .macro __v7_proc name, initfunc, mm_mmuflags = 0, io_mmuflags = 0, hwcaps = 0, proc_fns = v7_processor_functions
490         ALT_SMP(.long   PMD_TYPE_SECT | PMD_SECT_AP_WRITE | PMD_SECT_AP_READ | \
491                         PMD_SECT_AF | PMD_FLAGS_SMP | \mm_mmuflags)             
492         ALT_UP(.long    PMD_TYPE_SECT | PMD_SECT_AP_WRITE | PMD_SECT_AP_READ | \
493                         PMD_SECT_AF | PMD_FLAGS_UP | \mm_mmuflags)              
494         .long   PMD_TYPE_SECT | PMD_SECT_AP_WRITE | \                           
495                 PMD_SECT_AP_READ | PMD_SECT_AF | \io_mmuflags                   
496         initfn  \initfunc, \name                                                
497         .long   cpu_arch_name                                                   
498         .long   cpu_elf_name                                                    
499         .long   HWCAP_SWP | HWCAP_HALF | HWCAP_THUMB | HWCAP_FAST_MULT | \      
500                 HWCAP_EDSP | HWCAP_TLS | \hwcaps                                
501         .long   cpu_v7_name                                                     
502         .long   \proc_fns                                                       
503         .long   v7wbi_tlb_fns                                                   
504         .long   v6_user_fns                                                     
505         .long   v7_cache_fns                                                    
506 .endm                                        

매크로 변환후 

__v7_proc_info:
	.long   0x000f0000              @ Required ID value 
	.long   0x000f0000              @ Mask for ID
	.long	PMD_TYPE_SECT | PMD_SECT_AP_WRITE | PMD_SECT_AP_READ | PMD_SECT_AF | PMD_FLAGS_SMP
	.long PMD_TYPE_SECT | PMD_SECT_AP_WRITE | PMD_SECT_AP_READ | PMD_SECT_AF
	initfn	__v7_setup, __v7_proc
	.long   cpu_arch_name                                                   
	.long   cpu_elf_name                                                    
	.long   HWCAP_SWP | HWCAP_HALF | HWCAP_THUMB | HWCAP_FAST_MULT | HWCAP_EDSP | HWCAP_TLS
	.long   cpu_v7_name
	.long   v7_processor_functions
	.long   v7wbi_tlb_fns
	.long   v6_user_fns
	.long   v7_cache_fns
	.size   __v7_proc_info, . - __v7_proc_info
```