iamroot12CD/linux #34: 45주차 2016년 2월 27일 스터디

norux, 2016-02-27:
시간: 2016년 2월 27일 토요일 오후 3시 - 10시
장소: 강남 토즈타워점

rilts, 2016-02-27:
http://jake.dothome.co.kr/stext/

norux, 2016-02-27:
https://github.com/raspberrypi/linux/commit/72a20e22f49e2dad3180c23980a9df1c63faab0a

billpark, 2016-02-27:
#define DEFINE(sym, val) \
        asm volatile("\n->" #sym " %0 " #val : : "i" (val))

http://stackoverflow.com/questions/25294649/how-does-this-inline-assembly-define-a-variable

leemchaehoon, 2016-02-27:
inline 어셈블리 문서
https://wiki.kldp.org/KoreanDoc/html/EmbeddedKernel-KLDP/app3.basic.html

leemchaehoon, 2016-02-27:
리즈베리파이 vmlinux objectdump 결과
```
$ arm-linux-gnueabihf-objdump -d vmlinux |head -200

vmlinux:     file format elf32-littlearm

Disassembly of section .head.text:

80008000 <stext>:
80008000:	eb0055a6 	bl	8001d6a0 <__hyp_stub_install>
80008004:	e10f9000 	mrs	r9, CPSR
80008008:	e229901a 	eor	r9, r9, #26
8000800c:	e319001f 	tst	r9, #31
80008010:	e3c9901f 	bic	r9, r9, #31
80008014:	e38990d3 	orr	r9, r9, #211	; 0xd3
80008018:	1a000004 	bne	80008030 <stext+0x30>
8000801c:	e3899c01 	orr	r9, r9, #256	; 0x100
80008020:	e28fe00c 	add	lr, pc, #12
80008024:	e16ff009 	msr	SPSR_fsxc, r9
80008028:	e12ef30e 	.word	0xe12ef30e
8000802c:	e160006e 	.word	0xe160006e
80008030:	e121f009 	msr	CPSR_c, r9
80008034:	ee109f10 	mrc	15, 0, r9, cr0, cr0, {0}
80008038:	eb00052b 	bl	800094ec <__lookup_processor_type>
8000803c:	e1b0a005 	movs	sl, r5
80008040:	0a00053b 	beq	80009534 <__error>
80008044:	e28f3034 	add	r3, pc, #52	; 0x34
80008048:	e8930110 	ldm	r3, {r4, r8}
8000804c:	e0434004 	sub	r4, r3, r4
80008050:	e0888004 	add	r8, r8, r4
80008054:	eb00005c 	bl	800081cc <__vet_atags>
80008058:	eb000038 	bl	80008140 <__fixup_smp>
8000805c:	eb000009 	bl	80008088 <__create_page_tables>
80008060:	e59fd014 	ldr	sp, [pc, #20]	; 8000807c <stext+0x7c>
80008064:	e28fe00c 	add	lr, pc, #12
80008068:	e1a08004 	mov	r8, r4
8000806c:	e59ac010 	ldr	ip, [sl, #16]
80008070:	e08cc00a 	add	ip, ip, sl
80008074:	e1a0f00c 	mov	pc, ip
80008078:	ea00050b 	b	800094ac <__enable_mmu>
8000807c:	807752e0 	.word	0x807752e0
80008080:	80008080 	.word	0x80008080
80008084:	80000000 	.word	0x80000000

80008088 <__create_page_tables>:
80008088:	e2884902 	add	r4, r8, #32768	; 0x8000
8000808c:	e2444901 	sub	r4, r4, #16384	; 0x4000
80008090:	e1a00004 	mov	r0, r4
80008094:	e3a03000 	mov	r3, #0
80008098:	e2806901 	add	r6, r0, #16384	; 0x4000
8000809c:	e4803004 	str	r3, [r0], #4
800080a0:	e4803004 	str	r3, [r0], #4
800080a4:	e4803004 	str	r3, [r0], #4
800080a8:	e4803004 	str	r3, [r0], #4
800080ac:	e1300006 	teq	r0, r6
800080b0:	1afffff9 	bne	8000809c <__create_page_tables+0x14>
800080b4:	e59a7008 	ldr	r7, [sl, #8]
800080b8:	e28f0074 	add	r0, pc, #116	; 0x74
800080bc:	e8900068 	ldm	r0, {r3, r5, r6}
800080c0:	e0400003 	sub	r0, r0, r3
800080c4:	e0855000 	add	r5, r5, r0
800080c8:	e0866000 	add	r6, r6, r0
800080cc:	e1a05a25 	lsr	r5, r5, #20
800080d0:	e1a06a26 	lsr	r6, r6, #20
800080d4:	e1873a05 	orr	r3, r7, r5, lsl #20
800080d8:	e7843105 	str	r3, [r4, r5, lsl #2]
800080dc:	e1550006 	cmp	r5, r6
800080e0:	32855001 	addcc	r5, r5, #1
800080e4:	3afffffa 	bcc	800080d4 <__create_page_tables+0x4c>
800080e8:	e2840a02 	add	r0, r4, #8192	; 0x2000
[... 70 lines cut ...]
800081e4:	e3550005 	cmp	r5, #5
800081e8:	13550002 	cmpne	r5, #2
800081ec:	1a000004 	bne	80008204 <__vet_atags+0x38>
800081f0:	e5925004 	ldr	r5, [r2, #4]
800081f4:	e59f6014 	ldr	r6, [pc, #20]	; 80008210 <__vet_atags+0x44>
800081f8:	e1550006 	cmp	r5, r6
800081fc:	1a000000 	bne	80008204 <__vet_atags+0x38>
80008200:	e12fff1e 	bx	lr
80008204:	e3a02000 	mov	r2, #0
80008208:	e12fff1e 	bx	lr
8000820c:	edfe0dd0 	.word	0xedfe0dd0
80008210:	54410001 	.word	0x54410001

Disassembly of section .text:

80008240 <__turn_mmu_on>:
80008240:	e1a00000 	nop			; (mov r0, r0)
80008244:	f57ff06f 	isb	sy
80008248:	ee010f10 	mcr	15, 0, r0, cr1, cr0, {0}
8000824c:	ee103f10 	mrc	15, 0, r3, cr0, cr0, {0}
80008250:	f57ff06f 	isb	sy
80008254:	e1a03003 	mov	r3, r3
80008258:	e1a0300d 	mov	r3, sp
8000825c:	e1a0f003 	mov	pc, r3
```

minidump, 2016-02-27:
아까 스터디 진행중에 나왔던 sed관련 메모 추가
* sed를 이용해서 kbuild 헤더를 생성하는데 사용
* 대략 치환이 많은데 대부분 vi에서도 익숙한 정규 표현식을 사용
 * 정규 표현식을 사용하다보니 막강한 코드 생성기로 사용이 가능함
* 굳이 깊게 팔 필요는 없겠지만 적당한 튜토리얼은 이쪽 참고
 * http://www.grymoire.com/Unix/Sed.html

leemchaehoon, 2016-02-29:
45차 스터디를 정리한 PDF 문서 입니다.
[kernel_d_45.zip](https://github.com/iamroot12D/linux/files/150549/kernel_d_45.zip)

elinjkim, 2016-02-29:
[image]
[image]
[image]
[image]