iamroot12CD/linux #36: 46주차 2016년 3월 5일 스터디

norux, 2016-03-05:
시간: 2016년 3월 5일 토요일 오후 3시 - 10시
장소: 강남 토즈타워점

leemchaehoon, 2016-03-06:
46차 스터디 정리 문서 입니다.
[커널 12차 D팀 46차.pdf.zip](https://github.com/iamroot12D/linux/files/160166/12.D.46.pdf.zip)

rilts, 2016-03-08:
root@f96d53bd69ba:~/linux# arm-linux-gnueabihf-objdump -d vmlinux_VMSPLIT_1G | head -10

vmlinux_VMSPLIT_1G:     file format elf32-littlearm

Disassembly of section .head.text:

40008000 <stext>:
40008000:	eb0055be 	bl	4001d700 <__hyp_stub_install>
40008004:	e10f9000 	mrs	r9, CPSR
40008008:	e229901a 	eor	r9, r9, #26

root@f96d53bd69ba:~/linux# arm-linux-gnueabihf-objdump -d vmlinux_VMSPLIT_2G | head -10

vmlinux_VMSPLIT_2G:     file format elf32-littlearm

Disassembly of section .head.text:

80008000 <stext>:
80008000:	eb0054b6 	bl	8001d2e0 <__hyp_stub_install>
80008004:	e10f9000 	mrs	r9, CPSR
80008008:	e229901a 	eor	r9, r9, #26

root@f96d53bd69ba:~/linux# arm-linux-gnueabihf-objdump -d vmlinux_VMSPLIT_3G | head -10

vmlinux_VMSPLIT_3G:     file format elf32-littlearm

Disassembly of section .head.text:

c0008000 <stext>:
c0008000:	eb0055be 	bl	c001d700 <__hyp_stub_install>
c0008004:	e10f9000 	mrs	r9, CPSR
c0008008:	e229901a 	eor	r9, r9, #26

커널 영역을 1/2/3 기가로 나누도록 설정해서 커널 컴파일한 이미지를 덤프 해 봤습니다.

공부중에 나온 8000_0000 번지가 아무래도 커널영역 나누는 것과 가상메모리 시작 주소와 컴파일된 파일의 시작주소와 관계 있는 거 같아서 
arch/arm/configs/bcm2709_defconfig 파일의 CONFIG_VMSPLIT_xG=y 이렇게 설정변경해 가면서 커널 컴파일 했더니..  관계 있네요..