iamroot12CD/linux #37: 47주차 2016년 3월 12일 스터디

rilts, 2016-03-12:
시간: 2016년 3월 12일 토요일 오후 3시 - 10시
장소: 강남 토즈타워점

ahnchan2, 2016-03-12:
TTBCR
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0433a/CIHHACFF.html

rilts, 2016-03-12:
레지스터 2: 캐쉬설정레지스터

mrc p15, 0, Rd, c2, c0, 0 @ read data cacheable bits 
mcr p15, 0, Rd, c2, c0, 0 @ write data cacheable bits 
mrc p15, 0, Rd, c2, c0, 1 @ read instruction cacheable bits 
mcr p15, 0, Rd, c2, c0, 1 @ write instruction cacheable bits

leemchaehoon, 2016-03-12:
TTBR1
Translation Table Base Register 1

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0433a/CIHHACFF.html

minidump, 2016-03-12:
TTBR관련 iamroot 정리 문서
http://www.iamroot.org/xe/Kernel_10_ARM/176798

leemchaehoon, 2016-03-12:
Summary of CP15 instructions

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/CHDGIJFB.html

minidump, 2016-03-12:
UBFX(Unsigned Bit Field eXtract):

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489c/Cjahjhee.html
http://stackoverflow.com/questions/8366625/arm-bit-field-extract

minidump, 2016-03-12:
코프로세서15 관련 명령어 일체
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/CHDGIJFB.html

그중 Translation Table Base의 0번 레지스터 쓰기
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/BHCBIIGC.html

TTBCR
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/BIHGFCGF.html

leemchaehoon, 2016-03-12:
c10, Memory Region Remap Registers

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/CACHFIIH.html

rilts, 2016-03-12:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

ahnchan2, 2016-03-12:
PRRR 과 NMRR

http://www.iamroot.org/xe/Kernel_10_ARM/185229

leemchaehoon, 2016-03-12:
DSB

특수한 유형의 메모리 장벽 역할을 하는 데이터 동기화 장벽. 이 명령어가 완료될 때까지 프로그램 순서에서 이 명령어 뒤에 있는 명령어는 실행되지 않습니다. 이 명령어는 다음 경우에 완료됩니다.

    이 명령어 앞에 있는 모든 명시적 메모리 액세스가 완료될 경우 다음 사항이 적용됩니다.

    이 명령어 앞에 있는 모든 캐시, 분기 예측기 및 TLB 유지관리 작업이 완료될 경우 다음 사항이 적용됩니다.

허용되는 option의 값은 다음과 같습니다.

SY

    전체 시스템 DSB 작업. 기본값이며 생략할 수 있습니다.
ST

    저장이 완료될 때까지만 기다리는 DSB 작업
ISH

    공유 가능한 내부 도메인에만 수행되는 DSB 작업
ISHST

    저장이 완료될 때까지만 기다리고 공유 가능한 내부 도메인까지만 수행되는 DSB 작업
NSH

    통합 지점까지만 수행되는 DSB 작업
NSHST

    저장이 완료될 때까지만 기다리고 통합 지점까지만 수행되는 DSB 작업
OSH

    공유 가능한 외부 도메인에만 수행되는 DSB 작업
OSHST

    저장이 완료될 때까지만 기다리고 공유 가능한 외부 도메인까지만 수행되는 DSB 작업

leemchaehoon, 2016-03-12:
c1, Control Register

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/BABGDHIF.html

leemchaehoon, 2016-03-13:
TTBCR와 TTBR의 관계
[image]

slashdeva, 2016-03-13:
모임 사진 입니다.

[image]

[image]