iamroot12CD/linux #38: 47주차 2016년 3월 12일 스터디

minidump, 2016-03-12:
TTBR관련 iamroot 정리 문서
http://www.iamroot.org/xe/Kernel_10_ARM/176798

leemchaehoon, 2016-03-12:
BSYM 설명
http://www.iamroot.org/xe/Kernel_10_ARM/173070

leemchaehoon, 2016-03-12:
리즈베리파이 main id register
```
rpi1     rpi2
410FB767 410FC075 MRC p15, 0, r0, c0, c0, 0
1D152152 84448003 MRC p15, 0, r0, c0, c0, 1
00000000 00000000 MRC p15, 0, r0, c0, c0, 2
00000800 00000000 MRC p15, 0, r0, c0, c0, 3
410FB767 410FC075 MRC p15, 0, r0, c0, c0, 4
410FB767 80000F00 MRC p15, 0, r0, c0, c0, 5
410FB767 00000000 MRC p15, 0, r0, c0, c0, 6
410FB767 410FC075 MRC p15, 0, r0, c0, c0, 7
00000111 00001131 MRC p15, 0, r0, c0, c1, 0
00000011 00011011 MRC p15, 0, r0, c0, c1, 1
00000033 02010555 MRC p15, 0, r0, c0, c1, 2
00000000 00000000 MRC p15, 0, r0, c0, c1, 3
01130003 10101105 MRC p15, 0, r0, c0, c1, 4
10030302 40000000 MRC p15, 0, r0, c0, c1, 5
01222100 01240000 MRC p15, 0, r0, c0, c1, 6
00000000 02102211 MRC p15, 0, r0, c0, c1, 7
00140011 02101110 MRC p15, 0, r0, c0, c2, 0
12002111 13112111 MRC p15, 0, r0, c0, c2, 1
11231121 21232041 MRC p15, 0, r0, c0, c2, 2
01102131 11112131 MRC p15, 0, r0, c0, c2, 3
00001141 10011142 MRC p15, 0, r0, c0, c2, 4
00000000 00000000 MRC p15, 0, r0, c0, c2, 5
```

minidump, 2016-03-12:
UBFX(Unsigned Bit Field eXtract):  
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489c/Cjahjhee.html
http://stackoverflow.com/questions/8366625/arm-bit-field-extract

minidump, 2016-03-12:
코프로세서15 관련 명령어 일체
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/CHDGIJFB.html

그중 Translation Table Base의 0번 레지스터 쓰기
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360f/BHCBIIGC.html