iamroot12CD/linux #40: 3월 26일 스터디

mineenick, 2016-03-26:
장소 : 토즈 강남점
시작 시간 : 15:00

leemchaehoon, 2016-03-26:
문c 블로그
http://jake.dothome.co.kr/linux_kernel/linux_4/

ahnchan2, 2016-03-26:
Task, Process, Thread 차이
http://ljhh.tistory.com/entry/Task-Process-Thread

norux, 2016-03-26:
SCHED_FIFO  (RT) : 0 ~ 99
SCHED_RR     (RT) :  0 ~ 99
SCHED_NORMAL : 100 ~ 139
 
100을 기준으로 100 미만은 RT task,  100 이상은 일반 task로 구분

leemchaehoon, 2016-03-26:
동기화 기법: Read-Copy-Update
http://summerlight.tistory.com/entry/%EB%8F%99%EA%B8%B0%ED%99%94-%EA%B8%B0%EB%B2%95-Read-Copy-Update

norux, 2016-03-26:
http://www.valvers.com/programming/c/gcc-weak-function-attributes/

leemchaehoon, 2016-03-26:
Multiprocessor Affinity Register

The MPIDR characteristics are:

Purpose

    To identify:
        whether the processor is part of a Cortex-A9 MPCore implementation.
        Cortex-A9 processor accesses within a Cortex-A9 MPCore processor
        the target Cortex-A9 processor in a multi-processor cluster system.

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0388e/CIHEBGFG.html

ahnchan2, 2016-03-26:
set_my_cpu_offset() 함수 동작
http://www.iamroot.org/xe/index.php?mid=Kernel_10_ARM&document_srl=184082&sort_index=readed_count&order_type=desc