iamroot12CD/linux #42: 50주차 2016년 4월 2일 스터디

leemchaehoon, 2016-04-02:
시작합니다.
init/main.c

norux, 2016-04-02:
http://egloos.zum.com/studyfoss/v/5375570

leemchaehoon, 2016-04-02:
가변인자 호출
http://norux.me/19

slashdeva, 2016-04-02:
[image]
[image]
[image]
[image]

norux, 2016-04-02:
https://github.com/Rochester-NRT/AlphaGo

elinjkim, 2016-04-02:
[image]

mineenick, 2016-04-02:
TPPIDRPW에 대한 clue
https://lwn.net/Articles/527927/

leemchaehoon, 2016-04-02:
set_my_cpu_offset() 함수

http://www.iamroot.org/xe/index.php?mid=Kernel_10_ARM&document_srl=184082&sort_index=readed_count&order_type=desc

SehongKwon, 2016-04-02:
http://jake.dothome.co.kr/smp_setup_processor_id/
여기서 TPIDRPRW  검색

mineenick, 2016-04-02:
http://egloos.zum.com/studyfoss/v/5374731