iamroot12CD/linux #43: 51주차 2016년 4월 9일 스터디

leemchaehoon, 2016-04-09:
51주차 2016년 4월 9일 스터디

leemchaehoon, 2016-04-09:
라즈베리 파이 2 부팅 로그

http://fehead.nuxinfo.net/dokuwiki/doku.php?id=arm:raspberrypi:raspberrypi2_booting_log

leemchaehoon, 2016-04-09:
라즈베리 파이 레지스터 정보
http://fehead.nuxinfo.net/dokuwiki/doku.php?id=%EC%8A%A4%ED%84%B0%EB%94%94:arm_%EA%B4%80%EB%A0%A8_%EB%AC%B8%EC%84%9C

minidump, 2016-04-09:
asm 키워드에서 "memory"에 대해서,
확장 asm 문법의 마지막 인자인 clobbers의 경우 다음과 같이 2종류가 제공
 * "cc": condition code라고 해서 일종의 상태 레지스터에 해당
 * "memory": 일반적인 메모리 read/write에 해당
https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html#Clobbers

확장 asm 문법 다시보기
https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html

SehongKwon, 2016-04-09:
http://jake.dothome.co.kr/per-cpu/
per-cpu에 대하여 정리된 문C 블로그 주소.

minidump, 2016-04-09:
GCC의 빌트인 함수소개, 보통 __builtin_로 시작
분석중 등장한 __builtin_constant_p()의 경우, 컴파일타임에 상수인지 판별되면 1리턴, 반대는 0
https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html

SehongKwon, 2016-04-09:
arm builtin 함수들 한글설명
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0348ak/Cacdiihj.html

leemchaehoon, 2016-04-09:
라즈베리 파이2 config 파일

http://fehead.nuxinfo.net/dokuwiki/doku.php?id=arm:raspberrypi