iamroot12CD/linux #52: 58주차 2016년 5월 28일 스터디

leemchaehoon, 2016-05-28:
.bashrc 설정

```sh
export PS1='\e[0m[\e[1;36m\u\e[0m@\e[1;33m\h\e[0m \w \t]\n\$ '
export EDITOR=/usr/bin/vim

```

leemchaehoon, 2016-05-28:
IT expert 리눅스 디바이스 드라이버 소스

https://github.com/leemchaehoon/it_export_linux_device_driver

ahnchan2, 2016-05-28:
캐시 write 정책
http://talkingaboutme.tistory.com/482

leemchaehoon, 2016-05-28:
캐쉬 writeback alloc

https://encrypted.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=4&ved=0ahUKEwi8n82by_zMAhXlIaYKHZuNA4QQFggvMAM&url=http%3A%2F%2Fmpu.yonsei.ac.kr%2Fxe%2F%3Fmodule%3Dfile%26act%3DprocFileDownload%26file_srl%3D11170%26sid%3Dee4ab3e31f204e2a7804a59e9ce68727&usg=AFQjCNHnxCTS23ucZARkdTCDhJG_FFdQug&sig2=nP6Up3r4Ku2KbA5JRuKH2A&bvm=bv.123325700,d.dGY

mineenick, 2016-05-28:
http://egloos.zum.com/studyfoss/v/5259841

Geekjun, 2016-05-28:
http://ljhh.tistory.com/entry/Task-Process-Thread

ahnchan2, 2016-05-28:
VIPT, PIPT
http://manseok.blogspot.kr/2013_09_01_archive.html

mineenick, 2016-06-11:
http://egloos.zum.com/embedded21/v/530514